silvanmelchior/RPi_Cam_Web_Interface #696: Preview on the main screen

AdamKolany, 2024-06-16:
I have started the RPI_Cam_Web_Interface service on my Raspberry Pi. When I connect a monitor, the camera image overlays the desktop. However, when connecting via VNC, this image is not visible. 
switching off the camera in the web interface gets rid of the image. switching it on shows it again.

How can I remove the camera overlay without disabling the camera?

roberttidey, 2024-06-17:
Normally RPI_Cam_Web_Interface displays everything in a browser which can either be a browser on the raspberry or a remote browser on another platform.

There is a facility called hdmi_preview (off by default) for situations where one wants a local monitor on the raspberry to display a full resolution preview rather than use the desktop.

I suspect you have turned this on. The setting is hdmi_preview in the file /etc/raspimjpeg. Its value is 0 by default which means this is off.

ghengistom, 2024-06-25:
Tried running the ./install.sh on pi4 64 raspbian OS and got the following:

Package gpac is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'gpac' has no installation candidate
root@raspberrypi:/home/tom/RPi_Cam_Web_Interface#

Anybody run into this issue?