silvanmelchior/RPi_Cam_Web_Interface #698: Package 'gpac' has no installation candidate

ghengistom, 2024-06-25:
Tried running the ./install.sh on pi4 64 raspbian OS and got the following:

Package gpac is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'gpac' has no installation candidate
root@raspberrypi:/home/tom/RPi_Cam_Web_Interface#

Anybody run into this issue?

emjas92, 2024-06-25:
Yes, please check the existing discussion here https://github.com/silvanmelchior/RPi_Cam_Web_Interface/issues/689

kaihokori, 2024-10-26:
Hi @ghengistom, you might be interested in [this](https://github.com/silvanmelchior/RPi_Cam_Web_Interface/issues/688#issuecomment-2437456955). Be aware that not all functionality has been implemented but it should be a good starting point ^_^