silvanmelchior/RPi_Cam_Web_Interface #699: Suddenly all interface camera buttons and scheduling are not working

b1tfl0w, 2024-06-26:
Hello, I have been using this software for a year on a Raspberry pi Zero 2 with camera v1, I use it as a rearview mirror on my van, it is also configured as a security cam / dash cam, so I always keep motion capture enabled, I also have the scheduler set to change the exposure mode at night an so on. It always happened that every now and then within a day or two the cam would freeze and I would have to restart it but I had gotten used to this. Since last night, however, after restarting it after the usual freeze, all the buttons to record or start the motion detection no longer work, even the scheduling does not work, in fact the cam parameters have remained set as they were at the time of the freeze.
some info:
```
lsb_release -a
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 10 (buster)
Release:	10
Codename:	buster
```
/var/www/cam folder list:
```
total 1036
-rw-r--r-- 1 www-data www-data    119 Jun 26 00:41 cam_get.php
-rw-r--r-- 1 www-data www-data    471 Jun 26 00:41 cam_picLatestTL.php
-rw-r--r-- 1 www-data www-data   1200 Jun 26 00:41 cam_pic_new.php
-rw-r--r-- 1 www-data www-data    281 Jun 26 00:41 cam_pic.php
-rw-r--r-- 1 www-data www-data    698 Jun 26 00:41 cmd_func.php
-rw-r--r-- 1 www-data www-data    184 Jun 26 00:41 cmd_pipe.php
-rwxr-xr-x 1 www-data www-data  10106 Jun 26 00:41 config.php
-rw-r--r-- 1 www-data www-data     39 Jun 26 00:41 convertCmd.txt
-rw-r--r-- 1 www-data www-data   1284 Jun 26 00:41 Copyright.txt
drwxr-xr-x 2 www-data www-data   4096 Jul 14  2023 css
-rw-r--r-- 1 www-data www-data   2397 Jun 26 00:41 delete.png
-rw-r--r-- 1 www-data www-data     11 Jun 26 00:41 diskUsage.txt
-rw-r--r-- 1 www-data www-data      5 Jun 26 12:35 FIFO
prw-rw-rw- 1 www-data www-data      0 Jun 26 12:29 FIFO1
drwxr-xr-x 2 www-data www-data   4096 Jul 14  2023 fonts
-rw-r--r-- 1 www-data www-data   1078 Jun 26 00:41 image.png
-rw-r--r-- 1 www-data www-data  37452 Jun 26 00:41 index.php
drwxr-xr-x 2 www-data www-data   4096 Jul 14  2023 js
-rw-r--r-- 1 www-data www-data  25066 Jun 26 00:41 loading.jpg
drwxr-xr-x 2 www-data www-data   4096 Jun 25 21:02 macros
drwxr-xr-x 2 www-data www-data 671744 Jun 26 10:41 media
-rw-r--r-- 1 www-data www-data    346 Jun 26 00:41 min.php
-rw-r--r-- 1 www-data www-data   5966 Jun 26 00:41 motion.php
-rw-r--r-- 1 www-data www-data   1451 Jun 26 00:41 multiview.html
-rw-r--r-- 1 www-data www-data    134 Jun 26 00:41 multiview.jsonD
-rw-r--r-- 1 www-data www-data   1523 Jun 26 00:41 multiview.php
-rw-r--r-- 1 www-data www-data      1 Jun 26 00:41 pilight_off
-rw-r--r-- 1 www-data www-data      1 Jun 26 00:41 pipan_off
-rw-r--r-- 1 www-data www-data   3082 Jun 26 00:41 pipan.php
-rw-r--r-- 1 www-data www-data   1017 Jun 26 00:41 pirTrigger.py
-rw-r--r-- 1 www-data www-data  19985 Jun 26 00:41 preview.php
lrwxrwxrwx 1 www-data www-data     15 Jul 14  2023 raspimjpeg -> /etc/raspimjpeg
-rw-r--r-- 1 www-data www-data   1503 May 22 17:43 scheduleBackup.json
-rw-r--r-- 1 www-data www-data   1466 Jun 25 20:42 schedule.json
-rw-r--r-- 1 www-data www-data  20397 Jun 26 12:35 scheduleLog.txt
-rw-r--r-- 1 www-data www-data  38344 Jun 26 00:45 schedule.php
-rw-r--r-- 1 www-data www-data      0 Jun 26 00:41 servo_off
-rw-r--r-- 1 www-data www-data    258 Jun 26 00:41 status_mjpeg.php
lrwxrwxrwx 1 www-data www-data     31 Jun 26 00:44 status_mjpeg.txt -> /dev/shm/mjpeg/status_mjpeg.txt
-rw-r--r-- 1 www-data www-data   1417 Jun 26 00:41 timelapse.png
-rw-r--r-- 1 www-data www-data    247 Jun 26 12:29 uconfig
-rw-r--r-- 1 www-data www-data  27933 Jun 26 00:41 unavailable.jpg
-rw-r--r-- 1 www-data www-data  26210 Jun 26 00:41 updating.jpg
-rw-r--r-- 1 www-data www-data   1304 Jun 26 00:41 uPresetsN-IMX219.html
-rw-r--r-- 1 www-data www-data   1051 Jun 26 00:41 uPresetsN-OV5647.html
-rw-r--r-- 1 www-data www-data   1117 Jun 26 00:41 uPresetsP-IMX219.html
-rw-r--r-- 1 www-data www-data   1050 Jun 26 00:41 uPresetsP-OV5647.html
-rw-r--r-- 1 www-data www-data    449 Jun 26 00:41 uPresetsv2.html
-rw-r--r-- 1 www-data www-data    311 Jun 26 00:41 userbuttonsD
-rw-r--r-- 1 www-data www-data     23 Jun 26 00:41 userLevelExample
-rw-r--r-- 1 www-data www-data   1311 Jun 26 00:41 video.png
```

my schedule log:

```
[2024/06/26 00:14:01] RaspiCam support started
[2024/06/26 00:14:01] Capture Pipe already exists /var/www/html/FIFO1
[... 435 lines cut ...]
```
my schedule.json:
`{"Cmd_Poll":"0.03","Mode_Poll":"10","Management_Interval":"3600","Management_Command":"","PurgeVideo_Hours":"0","PurgeImage_Hours":"0","PurgeLapse_Hours":"0","GMTOffset":"2","PurgeSpace_ModeEx":"3","PurgeSpace_Level":"4","DawnStart_Minutes":"-20","DayStart_Minutes":"0","DayEnd_Minutes":"0","DuskEnd_Minutes":"20","Latitude":"41.78643742115127","Longitude":"12.41908564788736","Max_Capture":"0","DayMode":"0","AutoCapture_Interval":"0","AutoCamera_Interval":"0","Days":[["0","1","2","3","4","5","6"],["0","1","2","3","4","5","6"],["0","1","2","3","4","5","6"],["0","1","2","3","4","5","6"],["0","1","2","3","4","5","6"],["0","1","2","3","4","5","6"],["0","1","2","3","4","5","6"],["0","1","2","3","4","5","6"],["0","1","2","3","4","5","6"],["0","1","2","3","4","5","6"],["0","1","2","3","4","5","6"],["0","1","2","3","4","5","6"]],"Commands_On":["ca 1","ca 1","ca 1","ca 1","ca 1","","","","","","","","","","","",""],"Commands_Off":["ca 0","ca 0","ca 0","ca 0","ca 0","","","","","","","","","","","",""],"Modes":["","em night;mm average; px 1296  972  0.5  8 2592 1944 1; me 15; md 1; bu 300","em auto;mm spot; px 1296  972  25  25 2592 1944 1; me 150; md 1; bu 3000","em auto;mm average; px 1296  972  25  25 2592 1944 1; me 150; md 1; bu 3000","em auto;mm spot; px 1296  972  25  25 2592 1944 1; me 150; md 1; bu 3000","","","","","","","","","","","",""],"Times":["09:00","10:00","11:00","12:00","13:00","14:00","15:00","16:00","17:00","18:00","19:00","20:00"]}`

[image]

my uconfig file, which still has night settings even though it is now daytime, as if the scheduler does not update the data:
```
anno_background 1
rotation 180
quality 7
video_width 1296
video_height 972
video_fps 0.5
video_bitrate 5000000
video_buffer 3000
MP4Box_fps 8
vector_preview 0
motion_noise 1005
motion_threshold 10
motion_startframes 1
motion_clip 3
hdmi_preview 1

```

roberttidey, 2024-06-26:
Probably the quickest way to resolve this is to save your schedule settings file and your uconfig file then delete them. Restart the pi and make sure the interface is Ok under default conditions.

Then try introducing the uconfig file (stop the system, before copying and restart) and recheck. Then do the same with schedule.

This should tell you if it is associated with one of those two.

b1tfl0w, 2024-06-26:
After deleting those files and restarting the buttons are working again... but after I put the original files back (uconfig, schedule.json) the problem returns that the buttons don't work and the scheduler starts but doesn't seem to do anything.
I copied both files preseving permissions `-rw-r--r-- 1 www-data www-data` 
Before copying them I stopped the program with `./stop.sh` and restarted with `./start.sh`

b1tfl0w, 2024-06-26:
No sorry, I discovered that after deleting the files the buttons works only the first time are pressed, for example if I press "record image" the button turns to yellow for a sec, and I find the picture in the gallery, but if I press it a second time it does not, alse happens with "record video start" button, it turns yellow but if I try to press it again it won't stop, and I find the video in the gallery marked as busy.

roberttidey, 2024-06-26:
Ok.
maybe something has affected the permissions.

I suggest the following.

a) Stop the system
b) clear the media folder, uconfig,schedule settings (Save any stuff in media folder if you need it.)
c) run the ./install.sh again. This will refresh all the files and reset the permissions.

Then test using the defaults to make sure it is working OK.

roberttidey, 2024-06-27:
Good to hear all is OK.

The FIFO being in a different folder would explain things

If one needs to change the web folder, it is best to run the ./install.sh to do it as that ensures everything is pointing at the right places.