silvanmelchior/RPi_Cam_Web_Interface #707: Motion-detect dosn't trigger after moving the pi.

QuickyTank, 2025-02-02:
I setup a new Rpi-Cam-Web-interface for my daughter. Everything worked well and tested fine.
When I went to her house and installed it at her front door, it no longer triggers the motion record.
I've checked everything over and can't find any fault. I even put %c %f in the monitor time stamp, and turned on motion filter and everything works as it did before, but it doesn't trigger motion detect and therefor never starts recording. 
I setup this system for her becasue she had her Amazon parcels stolen a few times already. It records manually just fine. 
Please tell me what information you need from me to fix my problem.

roberttidey, 2025-02-03:
First thing to do is to check logging.

The best way to do this is to go into the schedule and clear the current logs.

Then perform an action that you believe should trigger motion detection.

Then check logs.

Note that when motion is detected then a signal is sent to the scheduler to decide what to do with it. So the schedule settings influence what actually happens when a motion detection signal is received.

The logs will show if a motion trigger happens and then what action if any is performed.

QuickyTank, 2025-02-04:
I'll try it this weekend when I go to her place.
What I can't figure out, is why it worked for a couple of days as I was
testing it to make sure when I took it there to install it, I won't have a
surprise.
Thank you for your excellent suggestion. If it doesn't work, I'll have to
reinstall the entire SD card.